stephendade/Rpanion-server #230: Is segmav possible to perform simple obstacle avoidance driving within the field of view

lida2003, 2024-05-29:
Hi @stephendade 

I have noticed your segmav, using semantic segmentation running on a Jetson Nano.
Is segmav possible to perform simple obstacle avoidance driving within the field of view?

I want to use it on my jetson orin nano board to drive my akm rover.

stephendade, 2024-05-29:
It doesn't do object avoidance. It does object tracking, where the rover will follow a specified segmentation class. So the opposite of what you want.

lida2003, 2024-05-29:
Actually, I need let rover do 3 things:

1. tracking the road
2. do simple object avoidance
3. following  a selected person

And this demo can do 1) , I'm looking for code to handle 1)2)3), do you have any suggestion?

stephendade, 2024-05-30:
No suggestion, sorry.

closed 2024-05-30 by lida2003