stephendade/Rpanion-server #250: 640x480 on RPI3B+ FPS can't be over 6.5, performance degraded

lida2003, 2024-09-28:
Current code on my RPI3B+ bookwrom system, FPS can't be more over 6.5.

```
Release date: July 4th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
```

The performance is significantly degraded, any way to boost?

lida2003, 2024-09-28:
I just used python code for test, which is much better.CPU is 300%, and it reached 22 FPS with software encoding. 

As I'm 64bit bookworn OS. It might be the same as [Pi4, which do not have hardware encorder "libOMX_Core.so"](https://github.com/raspberrypi/firmware/issues/1366).

```
import socket
import subprocess
import numpy as np
import picamera2
import time
import cv2  # Import OpenCV module

# Set up RTP streaming
rtp_host = "192.168.1.19"
rtp_port = 5400
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

# Use ffmpeg to encode video with specified profile and level
ffmpeg_command = [
    'ffmpeg',
    '-f', 'rawvideo',
    '-pix_fmt', 'yuv420p',  # Change to YUV 4:2:0 pixel format
    '-s', '640x480',  # Frame size
    '-r', '30',  # Frame rate
    '-i', '-',  # Input from stdin
    '-c:v', 'libx264',  # Use H.264 codec
    '-profile:v', 'baseline',  # Set profile to baseline
    '-level', '5.0',  # Set level to 5
    '-f', 'rtp',  # Output format
    f'rtp://{rtp_host}:{rtp_port}'
]

process = subprocess.Popen(ffmpeg_command, stdin=subprocess.PIPE)

camera = picamera2.Picamera2()
camera.configure(camera.create_video_configuration(main={"size": (640, 480)}))
camera.start()

try:
    while True:
        frame = camera.capture_array()
        # Convert frame to YUV 4:2:0
        frame_yuv = cv2.cvtColor(frame, cv2.COLOR_BGR2YUV_I420)
        process.stdin.write(frame_yuv.tobytes())  # Send the frame to ffmpeg
        time.sleep(0.033)  # Sleep for ~30 FPS
except KeyboardInterrupt:
    pass
finally:
    camera.stop()
    process.terminate()
    sock.close()
```

lida2003, 2024-09-28:
Tried 32bit bookworm, Not good!

lida2003, 2024-09-29:
with "high" profile, it reaches 25 ~27 FPS.

Two question:
1) I override frame rate to 60, it didn't improve any performance.   Is my configuration correct(two parameter changes)?
```
    '-r', '60',  # Frame rate
```
```
time.sleep(0.0167)  # Sleep for ~60 FPS
```

2) 32/64bit OS, I can't unlock omx hardware encorder, as it always prints below comments. I didn't find a way to get those omx files.

'''
[h264_omx @ 0x8358c0] libOMX_Core.so not found
[h264_omx @ 0x8358c0] libOmxCore.so not found
'''

stephendade, 2024-09-30:
> I'm NOT sure if rpanion(gst)/gstreamer is using the pi's hardware for encoding?

In the case of Bookworm, software encoding. There were some issues getting hardware encoding working (worked fine in Bullseye) so I had to force software encoding

Interesting that flipping the video appears to cause the performance issue. I'll look into that later in the week.

stephendade, 2024-10-07:
Yes, I can confirm this issue on Bookworm / RasPi3. I'll see if I can find a fix.

stephendade, 2024-10-10:
Can you try #253? That should help

lida2003, 2024-10-10:
I'll try when I have time. I think it might be tomorrow or day after tomorrow. Get back to you when I have got result.

lida2003, 2024-10-11:
@stephendade checked. It's good on my board, detailed info, please check PR [ Video: Use hardware encoding on < Rpi5 #253 ](https://github.com/stephendade/Rpanion-server/pull/253)

closed 2024-10-11 by lida2003