stephendade/Rpanion-server #257: Linting errors due to outdated or incomplete linter config

ddd999, 2024-10-18:
Running "npm run lint" on the current code in the repo returns 146 problems (143 errors, 3 warnings).  This is on a Raspberry Pi Zero 2W, if that makes any difference. As far as I can tell, most of these aren't actual problems but rather the linter not recognizing the syntax.

Just making the following change to the .eslintrc.json file reduces the output to 84 problems (81 errors, 3 warnings):

```
    "env": {
        "browser": true,
        "es6": true,
        "node": true,
        "mocha": true // add this
    },
```

However, that's still a lot of errors. I think a lot of (or potentially all of) those are due to issues with the linter config. For example, ten of those are: `Parsing error: Unexpected token =` for function declarations in the format `functionName= (value, action) => {`.

Based on my Google/StackOverflow searches it seems like the updated @babel/eslint-parser is needed, but either the version of eslint used by Rpanion is too old, or there is some other configuration issue that I haven't been able to sort out. Here are my modified config files:

[package.json](https://github.com/user-attachments/files/17427611/package.json)
[.eslintrc.json](https://github.com/user-attachments/files/17427612/default.eslintrc.json)
[babelrc.txt](https://github.com/user-attachments/files/17427616/babelrc.txt)

So far I haven't been able to get any further than that without descending into full blown npm dependency hell. Happy to test out any suggestions!

stephendade, 2024-10-18:
Thanks for checking. With eslint, in my recent patch (https://github.com/stephendade/Rpanion-server/commit/8c47dbd1f76dca8c914276435a61afde20635220#diff-7ae45ad102eab3b6d7e7896acd08c427a9b25b346470d7bc6507b6481575d519) I needed to leave eslint un-upgraded due to a bunch of dependency issues.

When I've got some time in the next week or two, I'll dig in a bit more.

stephendade, 2024-10-27:
I've got a basic upgraded ESLint working in the above PR. Now I just have to fix up all the linting errors :)

stephendade, 2024-11-11:
Fixed in #258

closed 2024-11-11 by stephendade