stephendade/Rpanion-server #268: Packets but no messages

Steven1994, 2025-01-17:
I've setup a pi zero (version 0.8 of the server, highest version available for pi zero 1) connected to a pixhawk 2 cube black.

I've set baud rate to 57600 and Mavlink version to 2. This is the same on my pixhawk and in the rpanion server. I can see that I get packages but there a no messages displayed in the console.

This result in that if I connect mission planner via udp or tcp it gets stuck on getting parameters...

Versions:
Server 0.8
Ardupilot copter 4.5.7

What could be the issue that I don't get the correct messages? 

[image]

stephendade, 2025-01-18:
Can you check the CPU and RAM usage on the Pi?

I do recall that I ran into some issues with the Pi Zero1 running out of CPU/RAM in some cases.

Steven1994, 2025-01-18:
Hi, thanks for the quick reply, i just made some progress. I got it partially working by changing the following parameters from 0 to these values. Now i have my virtual horizon working for example. Only the full parameter is still not being loaded and the box with 'getting parameters' keeps staying on the screen. 

[image]

Steven1994, 2025-01-18:
I got it working, serial2_options was set to 4 instead of 0, so it was half duplex.