stephendade/Rpanion-server #270: Unable to load Wireguard key

mishinroman94, 2025-01-25:
[image]

mishinroman94, 2025-01-25:
I had to manually upload the key file via SSH. In this case, everything works and is displayed, but it is not possible via the web interface after the last update. I think that the problem may be related to the authorization mod

stephendade, 2025-01-26:
Thanks for the report ... I'll look into it

closed 2025-06-22 by stephendade

Autonomost, 2025-08-14:
@mishinroman94 where did you place the key file? I'm having issues with this too. 

@stephendade I'm using the latest image and still having this issue - even tried updating via debian and the web gui. no luck. The version number doesn't change. Want me to make a separate issue?

stephendade, 2025-08-15:
@Autonomost: As in my previous post, it's fixed in the deb package (not the disk image). I hope to have an updated disk image with the fix in the next few months.