stephendade/Rpanion-server #282: Add support for Mission Planner video view

stephendade, 2025-04-12:
Mission Planner now has a video stream auto-discovery (https://discuss.ardupilot.org/t/mission-planner-live-video-support/132920). Currently doesn't appear to work with Rpanion, so need to investigate further.

-Mission Planner does query Rpanion for ``CameraInformation`` and gets the correct response
-Mission Planner does _not_ query Rpanion for ``VideoStreamInformation``, which has the actual stream details.

ddd999, 2025-04-14:
This ties in to the updates to the photo/video system that I'm working on in PR #229, which I realize now that got auto closed because I deleted and recreated the branch to fix a git mess on my end. Is it possible to reopen that PR?

Hoping to spend some more time on it this week.

stephendade, 2025-04-14:
Cool - great to know!

Can't re-open the PR from my end. You might need to rename the branch to a new name and open a new PR.