stephendade/Rpanion-server #283: Request for Radxa Zero 3 Support

shutongwu, 2025-04-17:
Could you consider adding firmware support for Radxa Zero 3? This device has hardware H.265 encoding capability. The current Zero 2W struggles significantly with 1080p@30fps streaming (severe frame drops), and lowering resolution triggers a center-cropping behavior instead of proper scaling. Additionally, H.265 support would greatly benefit bandwidth-limited scenarios like 4G networks (even though I'm currently on 5G, network conditions can vary).

stephendade, 2025-04-19:
That is quite a cheap board for hardware x265 encoding.

Happy to accept PR's to support that board :)

shutongwu, 2025-04-22:
Sorry, I previously said that Radxa was excellent. My positive impression was based on my H265 streaming tests, but that was when I was always flying missions where I didn’t care about latency (latency was set to 100ms). It wasn’t until I started testing FPV flying with latency=0 these past few days that I realized the real situation. mpph265enc is just a pile of crap. v4l2h264enc is still the best choice.

Sorry for wasting your time replying to me. I don’t think I’ll continue using the Radxa Zero3W. But I’ll still share my experience for reference:
If you don’t care about latency (buffering delay greater than 80ms) and want high image quality, then using mpph265enc on the Radxa Zero3W is fine.
However, the v4l2h264enc on Raspberry Pi performs much better in situations with network jitter or instability (especially with 0 latency settings).

stephendade, 2025-04-22:
That's quite interesting. Thanks for investigating.

I have found that, in some cases, the hardware x264/5 encoders do not have all the features required for low latency realtime streaming. This appears to be one such case.