stephendade/Rpanion-server #319: Videostream problem with 0.11 on Raspberrypi 4 and cm4

AndreasHIGHCAT, 2025-08-27:
Using the 0.11 Raspberry Pi 3, 4, Zero2 image (https://github.com/stephendade/Rpanion-server/releases/download/v0.11.0/Rpanion-Server-0.11.0-RasPiOS.img.xz)
I can do the settings

[image]

The RTSP stream is reachable for 1 second then frezes/breaks down.

this cant be a setup problem as we are connected directly via ethernet to the raspberrypi and the same setup is working with the Rapberrry pi 5 and the setup is funtional with the 0.10 Release with a cm4

stephendade, 2025-08-29:
The latest master (or V11.4) have improvements for video handling. Can you check if these work for you?

stephendade, 2025-09-06:
> With the release V11.4? do i need to install Mavlink-router myself or should it install with that? i get the error code that Mavlink_routerd is missing in the path

Yes, for 11.4 you need to install mavlink-router manually. For future versions (and current Github master), mavlink-router will be included in the installation.

> then there is no service created which was previosly the case

Correct. I'll need to change the names of the deploy scripts, as they no longer do deployment. There's been a significant change in how Rpanion-server is installed as it's now a packaged ``deb`` archive. I will need to update the documentation - thanks for reporting this.

stephendade, 2025-10-08:
I've now go some time to look at this. I can confirm the issue (on master) on the Raspi4 and CM4. I _think_ it's an issue with how the hardware encoder is called.

I should have a fix in the next few days.